nimaltd/ee #3: Returns false, Please add example

pooyasa, 2020-01-15:
Hi,
first of all, thank you for this library
I wanted to use this to store permanent data on my STM32F030K6T6
However,` EE_Write()` and `EE_Writes()` both return false on 
`if(HAL_FLASH_Program(FLASH_TYPEPROGRAM_WORD((i+StartVirtualAddress)*4)+_EEPROM_FLASH_PAGE_ADDRESS,(uint64_t)Data[i])!=HAL_OK)
`
which is on line 172 of eeprom.c

nimaltd, 2020-01-15:
Do you config correctly eepromconf.h ?Send to me

hosseinghaheri, 2022-07-18:
Hi. I have the same problem with STM32F030K6T6.
The ee_write function returns false in all cases.

If the microcontroller is programmed in Erase Full Chip mode.
The ee_write function returns false, but the storage is done.

Pay attention to the following code (Erase Full Chip mode):

	uint8_t to_write = 125;
	uint8_t to_read = 0;
	
	ee_init();
	
	ee_write(0,1, &to_write);            // return False
	ee_read(0,1, &to_read);              // to_read is 125
	
	to_write=150;
	
	ee_write(0,1, &to_write);            // return False
	ee_read(0,1, &to_read);              // to_read is 125