nimaltd/ee #4: STM32G0

comato, 2020-02-29:
Nice library... How to use it on STM32G0?

nimaltd, 2020-02-29:
Hello. I did not test on G series. But I think work.

comato, 2020-02-29:
I tried it with STM32G030 (32KB flash) But in HAL are some differencies,
FLASH_TYPEPROGRAM_WORD is changed to FLASH_TYPEPROGRAM_DOUBLEWORD
flashErase.PageAddress is changed to flashErase.Page
Also I am not sure what is page size (2048 bytes?)

comato, 2020-02-29:
it is working, to flashErase.Page is needed to write page number, not addres. Also I changed all /4 *4 to /8 *8 because DOUBLEWORD ...

closed 2020-02-29 by comato

leech001, 2022-09-17:
> it is working, to flashErase.Page is needed to write page number, not addres. Also I changed all /4 *4 to /8 *8 because DOUBLEWORD ...

Hi. Can you share your config for STM32G030F6P6?

leech001, 2023-01-01:
Added ability to work with STM32G030xx series #18