nimaltd/ee #5: Writing Half word (16-bit) instead of Word(32-bits)

HamzaHajeir, 2020-03-04:
Hi

What would happen if I decided to store 16-bits variables instead of 32-bits ? Would that free up spaces ? As it's mentioned in [ST EEPROM Emulation Docs](https://www.st.com/content/ccc/resource/technical/document/application_note/ee/ef/d7/87/cb/b7/48/52/CD00165693.pdf/files/CD00165693.pdf/jcr:content/translations/en.CD00165693.pdf)

Or it's prohibited ?

Also, Can you provide an example to know where to start ? ST eeprom emulation has an example and clear way to use that lib. (EE_init() is the starter ofcourse.) But here it's confusing and don't know where to start, Don't know how it work, or how should it work. As whether it label pages in headers or not..

Thanks

nimaltd, 2020-03-04:
Hello. You can convert any data type to uint32 array by memcpy function.

HamzaHajeir, 2020-03-05:
Thanks, 

How about the straight use of the library, Can you provide a demo example ?

nimaltd, 2020-03-05:
I dont have yest. But its simple.

michelkeijzers, 2020-05-08:
First, thank you for creating this EEPROM library. I'm using it and it works. However, I'm wondering why  in eepromConf.h,  _EEPROM_USE_FLASH_PAGE is a define/constant. If it would be a variable, than some wear level machinism could be built in... is there a specific reason for it being constant?

nimaltd, 2020-05-09:
@michelkeijzers  . waiting for next version :)

michelkeijzers, 2020-05-09:
> @michelkeijzers . waiting for next version :)

I already made changes, but in my own git repository https://github.com/michelkeijzers/FuzzTester2 ... Note however, I removed also a bit of code (for STMF0 since I will not use it), and I added some variables and extra functions, which might slow down (slightly) performance and memory usage.