nimaltd/ee #7: Use of this lib for stm32l476 in shutdown mode

axoulc, 2021-02-25:
Hello and thanks for your library.
I tried to adapt it to L476 with the following parameters :
```C
#define   _EE_USE_FLASH_PAGE_OR_SECTOR              (255)
...
#if defined(STM32L476xx)
#define   _EE_SIZE              2048
#define   _EE_ADDR_INUSE        (((uint32_t)0x08000000) | (_EE_SIZE * _EE_USE_FLASH_PAGE_OR_SECTOR))
#define   _EE_FLASH_BANK        FLASH_BANK_1
#define   _EE_PAGE_OR_SECTOR    PAGE_NUM
#if (_EE_USE_FLASH_PAGE_OR_SECTOR > 255)
#error  "Please Enter correct address, maximum is (255)"
#endif
#endif
```
It works !
But I need to use your lib in order to backup some data to enter in shutdown mode. I had the following idea in a sort of algorithm :

- First boot : `ee_init` and `ee_erase`
- Before shutdown : write : first byte is the number of data written just after. Next bytes : data
- Sleep
- After shutdown : `ee_init` , read the first byte to know the size of the next reading. Read the following bytes according to the size
- Before shutdown : rewrite the first byte to the following append. Write after existing bytes new data.

Is this algorithm is possible according to your lib and flash caracteristics ?
Thanks in advance and sorry for my english
Axoul

michelkeijzers, 2021-03-02:
Dear Axoul,

To be honest, I don't know if it works. I never tried anything with sleeping mode, so you would have to try yourself I'm afraid.
I'm glad at least you easily could port it to your L476.

Kind regards,

Michel

nimaltd, 2021-03-05:
hi. do you execute ee_commit() ? you can write your data by ee_writeToRam() and ee_commit .

axoulc, 2021-03-23:
Hi, I did my first test and it almost works.
I write to FLASH 16 bytes (struct) and I rebuild the struct just after a complete shutdown.
Here my code :
```C
#define NBDATA 2

typedef struct {
	time_t timestamp;
	float data[NBDATA];
} __attribute__((__packed__)) loggerData_t;
```
```C
  ee_init();
  uint8_t buffer[sizeof(loggerData_t)];
  if (startFromShutdown == 0) {
	  uint8_t size = 3;
	  loggerData_t testdata[3];
	  testdata[0] = (loggerData_t){1616494884, {26.54, 132.98}};
	  testdata[1] = (loggerData_t){1616494885, {27.96, 1.365}};
	  testdata[2] = (loggerData_t){1616494886, {28.563, 45.4545}};
	  ee_format(false);
	  ee_writeToRam(0, 1, &size);

	  for (uint8_t i = 0, j = 1; i < 3; i++, j+=sizeof(loggerData_t)) {
		  memcpy(buffer, &testdata[i], sizeof(loggerData_t));
		  ee_writeToRam(j, sizeof(loggerData_t), buffer);
	  }
	  ee_commit();
  } else {
	  uint8_t datanb;
	  ee_read(0, 1, &datanb);
	  for (uint8_t i = 0, j = 1; i < datanb; i++, j+=sizeof(loggerData_t)) {
		  ee_read(j, sizeof(loggerData_t), buffer);
		  memcpy(&received[i], buffer, sizeof(loggerData_t));
	  }
  }
  shutdownTimer(20); //Shutdown
```
All variables of the struct are good execpt one :
[image]
Write
[image]
Read

It appears for the three struct rebuilding, always on the same bit.
If anyone could help me, it'll be awesome.
Axoul ;)

nimaltd, 2021-03-24:
ee_init();
ee_read(0,sizeof(yourstruct),(uint8_t*)&yourstruct);
...
Change your struct
...
ee_writetoram(0,sizeof(yourstruct),(uint8_t*)&yourstruct);
ee_commit();

axoulc, 2021-03-25:
Hi, thanks for your answer but the problem still the same. My code :
```C
typedef struct __packed {
	time_t timestamp;
	float temp;
	float rand;
} loggerData_t;
```
```C
if(!startFromShutdown) {
	RV3032_SetTimeDate(&rtc, 1616577571);
	eepromInit();
	RV3032_GetTemperature(&rtc);
	RV3032_GetTimeDate(&rtc);
	inData = (loggerData_t){rtc.lastEpoch, rtc.lastTemp, randFloat()};
	printf("%f|%f\r\n", inData.temp, inData.rand); //debug
	eepromAppend(&inData);
	shutdownTimer(10);
} else {
	if (eepromNbData() == 5) {
		eepromToFlash();
	} else {
		RV3032_GetTemperature(&rtc);
		RV3032_GetTimeDate(&rtc);
		inData = (loggerData_t){rtc.lastEpoch, rtc.lastTemp, randFloat()};
		printf("%f|%f\r\n", inData.temp, inData.rand); //debug
		eepromAppend(&inData);
		shutdownTimer(10);
	}
}
```
```C
void eepromInit(void) {
	uint8_t nb = 0;
	ee_init();
	ee_format(false);
	ee_write(0, 1, &nb);
}

void eepromAppend(loggerData_t *newData) { // max 512 bytes (heap)
	uint8_t nbData = eepromNbData();
	uint8_t newNum = nbData + 1;
	ee_writeToRam(0, 1, &newNum);
	for (uint8_t i = 0, j = 1; i < nbData; i++, j += sizeof(loggerData_t)) {
		loggerData_t buffer;
		ee_read(j, sizeof(loggerData_t), (uint8_t*) &buffer);
		ee_writeToRam(j, sizeof(loggerData_t), (uint8_t*) &buffer);
	}
	ee_writeToRam((1 + (nbData * sizeof(loggerData_t))), sizeof(loggerData_t),
			(uint8_t*) newData);
	ee_commit();
}

uint8_t eepromNbData(void) {
	uint8_t nbData = 0;
	ee_read(0, 1, &nbData);
	return nbData;
}

void eepromToFlash(void) {
	uint8_t nbData = eepromNbData();
	for (uint8_t i = 0, j = 1; i < nbData; i++, j += sizeof(loggerData_t)) {
		loggerData_t out;
		ee_read(j, sizeof(loggerData_t), (uint8_t*) &out);
		char uartBuffer[30];
		sprintf(uartBuffer, "%ld,%.3f,%.3f\r\n", out.timestamp, out.temp,
				out.rand);
		HAL_UART_Transmit(&huart2, (uint8_t*) uartBuffer, strlen(uartBuffer),
				1000);
	}
	eepromInit();
}
```
This program append 5 times the struct to the existing data. After that it prints it :
```
24.125000|1.400051 // data 1
24.000000|4.350108 // data 2
23.812500|2.300165 // data 3
23.625000|0.250222 // data 4
23.500000|3.200278 // data 5
1616577571,0.024,1.400 //Final print
1616577582,0.094,4.350
1616577593,0.372,2.300
1616577604,1.477,0.250
1616577615,5.875,3.200
```
Thanks in advance
Axoul

nimaltd, 2021-03-27:
try this one
```
typedef struct  {
	float temp;
	float rand;
	time_t timestamp;

} loggerData_t;
```