nimaltd/ee #8: Configs for STM32F411

VoidSeeder, 2021-03-03:
Hello @nimaltd 

I am a beginner in STM32 and would like to know the information's that I must add in ee.c to use the library in STM32F411. The information I would like to know are the parameters:

_EE_SIZE
_EE_ADDR_INUSE
_EE_FLASH_BANK
_EE_VOLTAGE_RANGE
_EE_PAGE_OR_SECTOR

Thanks for listening!
João Pedro Vilela

nimaltd, 2021-03-04:
Hello. I think the same of 407 series
add this code  and try.

#if defined(STM32F411xx)
#define   _EE_SIZE              (1024 * 128)
#define   _EE_ADDR_INUSE        (((uint32_t)0x08020000) | (_EE_SIZE*(_EE_USE_FLASH_PAGE_OR_SECTOR - 5)))
#define   _EE_FLASH_BANK        FLASH_BANK_1
#define   _EE_VOLTAGE_RANGE     _EE_VOLTAGE
#define   _EE_PAGE_OR_SECTOR    PAGE
#if (_EE_USE_FLASH_PAGE_OR_SECTOR > 11)
#error  "Please Enter correct address, maximum is (11)"
#endif
#if (_EE_USE_FLASH_PAGE_OR_SECTOR < 5)
#error  "Please Enter correct address, minimum is (5)"
#endif
#endif

VoidSeeder, 2021-03-04:
Thanks for the answer!

I tried it and have some build errors related to page or sector. Seems like F411 have sectors. Can I just change this macro value? Can I have bigger problems if I just try to change these values ​​and run the debug?

Build return of this try:

> 13:10:13 **** Incremental Build of configuration Debug for project newWifi ****
make -j4 all 
arm-none-eabi-gcc "../Drivers/ee/ee.c" -mcpu=cortex-m4 -std=gnu11 -g3 -DUSE_HAL_DRIVER -DSTM32F411xE -DDEBUG -c -I../Core/Inc -I../Drivers/STM32F4xx_HAL_Driver/Inc -I../Drivers/STM32F4xx_HAL_Driver/Inc/Legacy -I../Drivers/CMSIS/Device/ST/STM32F4xx/Include -I../Drivers/CMSIS/Include -I../Middlewares/Third_Party/FreeRTOS/Source/include -I../Middlewares/Third_Party/FreeRTOS/Source/CMSIS_RTOS -I../Middlewares/Third_Party/FreeRTOS/Source/portable/GCC/ARM_CM4F -O0 -ffunction-sections -fdata-sections -Wall -fstack-usage -MMD -MP -MF"Drivers/ee/ee.d" -MT"Drivers/ee/ee.o" --specs=nano.specs -mfpu=fpv4-sp-d16 -mfloat-abi=hard -mthumb -o "Drivers/ee/ee.o"
arm-none-eabi-gcc "../Drivers/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_flash_ramfunc.c" -mcpu=cortex-m4 -std=gnu11 -g3 -DUSE_HAL_DRIVER -DSTM32F411xE -DDEBUG -c -I../Core/Inc -I../Drivers/STM32F4xx_HAL_Driver/Inc -I../Drivers/STM32F4xx_HAL_Driver/Inc/Legacy -I../Drivers/CMSIS/Device/ST/STM32F4xx/Include -I../Drivers/CMSIS/Include -I../Middlewares/Third_Party/FreeRTOS/Source/include -I../Middlewares/Third_Party/FreeRTOS/Source/CMSIS_RTOS -I../Middlewares/Third_Party/FreeRTOS/Source/portable/GCC/ARM_CM4F -O0 -ffunction-sections -fdata-sections -Wall -fstack-usage -MMD -MP -MF"Drivers/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_flash_ramfunc.d" -MT"Drivers/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_flash_ramfunc.o" --specs=nano.specs -mfpu=fpv4-sp-d16 -mfloat-abi=hard -mthumb -o "Drivers/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_flash_ramfunc.o"
arm-none-eabi-gcc "../Drivers/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_gpio.c" -mcpu=cortex-m4 -std=gnu11 -g3 -DUSE_HAL_DRIVER -DSTM32F411xE -DDEBUG -c -I../Core/Inc -I../Drivers/STM32F4xx_HAL_Driver/Inc -I../Drivers/STM32F4xx_HAL_Driver/Inc/Legacy -I../Drivers/CMSIS/Device/ST/STM32F4xx/Include -I../Drivers/CMSIS/Include -I../Middlewares/Third_Party/FreeRTOS/Source/include -I../Middlewares/Third_Party/FreeRTOS/Source/CMSIS_RTOS -I../Middlewares/Third_Party/FreeRTOS/Source/portable/GCC/ARM_CM4F -O0 -ffunction-sections -fdata-sections -Wall -fstack-usage -MMD -MP -MF"Drivers/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_gpio.d" -MT"Drivers/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_gpio.o" --specs=nano.specs -mfpu=fpv4-sp-d16 -mfloat-abi=hard -mthumb -o "Drivers/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_gpio.o"
arm-none-eabi-gcc "../Drivers/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_pwr.c" -mcpu=cortex-m4 -std=gnu11 -g3 -DUSE_HAL_DRIVER -DSTM32F411xE -DDEBUG -c -I../Core/Inc -I../Drivers/STM32F4xx_HAL_Driver/Inc -I../Drivers/STM32F4xx_HAL_Driver/Inc/Legacy -I../Drivers/CMSIS/Device/ST/STM32F4xx/Include -I../Drivers/CMSIS/Include -I../Middlewares/Third_Party/FreeRTOS/Source/include -I../Middlewares/Third_Party/FreeRTOS/Source/CMSIS_RTOS -I../Middlewares/Third_Party/FreeRTOS/Source/portable/GCC/ARM_CM4F -O0 -ffunction-sections -fdata-sections -Wall -fstack-usage -MMD -MP -MF"Drivers/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_pwr.d" -MT"Drivers/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_pwr.o" --specs=nano.specs -mfpu=fpv4-sp-d16 -mfloat-abi=hard -mthumb -o "Drivers/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_pwr.o"
../Drivers/ee/ee.c:136:2: error: #error "Please Enter correct address, maximum is (11)"
 #error "Please Enter correct address, maximum is (11)"
  ^~~~~
../Drivers/ee/ee.c: In function 'ee_format':
../Drivers/ee/ee.c:163:12: error: 'FLASH_EraseInitTypeDef {aka struct <anonymous>}' has no member named 'NbPages'
  flashErase.NbPages = 1;
            ^
../Drivers/ee/ee.c:164:13: error: 'FLASH_EraseInitTypeDef {aka struct <anonymous>}' has no member named 'PageAddress'
   flashErase.PageAddress = _EE_ADDR_INUSE;
             ^
../Drivers/ee/ee.c:165:26: error: 'FLASH_TYPEERASE_PAGES' undeclared (first use in this function); did you mean 'TYPEERASE_PAGES'?
   flashErase.TypeErase = FLASH_TYPEERASE_PAGES;
                          ^~~~~~~~~~~~~~~~~~~~~
                          TYPEERASE_PAGES
../Drivers/ee/ee.c:165:26: note: each undeclared identifier is reported only once for each function it appears in
make: *** [Drivers/ee/subdir.mk:18: Drivers/ee/ee.o] Error 1
make: *** Waiting for unfinished jobs....
"make -j4 all" terminated with exit code 2. Build might be incomplete.
13:10:14 Build Failed. 6 errors, 0 warnings. (took 730ms)

Thanks for listening!
João Pedro Vilela

nimaltd, 2021-03-05:
I update it. Download again

closed 2021-03-05 by nimaltd

VoidSeeder, 2021-03-05:
Thanks! This will help a lot.