nimaltd/ee #12: FE instead of FF

ednspace, 2022-02-25:
Hello,
I am using a STM32L476RG.

//From eeConfig.h
#define   _EE_PAGE_OR_SECTOR				(2)          //For page writes instead of by address
#define   _EE_USE_FLASH_PAGE_OR_SECTOR         (255)      //Last page of bank 1
#define   _EE_USE_RAM_BYTE                                 (1048)    //Have no idea ;)

//From ee.c
#if defined(STM32L476xx)
#define   _EE_SIZE              2048
#define   _EE_ADDR_INUSE        (((uint32_t)0x08000000) | (_EE_SIZE * _EE_USE_FLASH_PAGE_OR_SECTOR))
#if (_EE_USE_FLASH_PAGE_OR_SECTOR < 256)
#define   _EE_FLASH_BANK        FLASH_BANK_1
#endif

#if (_EE_USE_RAM_BYTE > 0)
uint8_t ee_ram[_EE_USE_RAM_BYTE];
#endif

For the most part everything is working as expected except when I write the data into cleared FLASH some of 64 bit double words end up as  FFFFFFFE instead of FFFFFFFF. This corruption occurs even in areas I am not writing. 

I am sure its likely the way I have your library configured however I have been struggling for a few days now and so far I have not been able to solve it.

This is what I am writing. First I read in the flash and print via serial, its during debug and I have mass erase enabled in the IDE so the first set is all FF's next I write my buffer which is 25 words long starting with HEX 7C so far so good

[image]

after this write and commit here is a picture of the flash memory as reported by the IDE

[image]

You can see the FE in almost every other word except for some unknown reason in the first two. Also notice how the FE corruption extends off into FLASH even into areas I am not writing. It extends down a little over 1k. These are all showing as FF's before I write and commit and then change to FE's after

To make matters more interesting, in the area that I am writing the FE's almost always corresponds to me trying to write an FF. 
I know crazy right ;) 

I love the your library and it seems like a perfect low overhead way to manage a couple of configuration buffers for my project.
Its really pretty perfect if I could just make it work ;)
Thanks for any help, 
Eric

nimaltd, 2022-02-25:
Hi 
Never use page 2. You have to use the last of sector or page on your device

ednspace, 2022-02-25:
thanks for getting back to me.
I thought I was using page 255 - 0x0807F800
I think this is the last page of Bank 1 on this device.

Which setting is showing that I am using page 2 ?

ednspace, 2022-02-25:
The define
#define _EE_PAGE_OR_SECTOR (2) //For page writes instead of by address
I set that to 2 so that hopefully it chooses PAGE_NUM writes instead of PAGE 
//From ee.c
#define   PAGE                  0
#define   SECTOR                1
#define   PAGE_NUM              2

That seemed to work and turns off flashErase.PageAddress which does not seem to be supported by this processor