nimaltd/ee #13: STM32F411xE - use lower sector

magdkudama, 2022-03-04:
Hello,

Thanks for this library.

Just a quick question: why do you only allow writing on sectors 5-7 on the device, knowing they're 128Kb each (in my case)? Is it a bad practice to use the lower sectors (say, sector 1, which is 16Kb)?

Thanks!

nimaltd, 2022-03-04:
Hello. You have to choose the last sector  because your code program into sector 0 to the last

magdkudama, 2022-03-04:
Thanks for the comment. I guess that's not always true? I have a bootloader on sector 0, and it decides which app will be used (I can have two apps). Both apps live on the higher end of sectors, as they're bigger. So for me it's more interesting to have my data saved in a lower sector. Is that not a valid use-case?

Thanks a lot.

nimaltd, 2022-03-05:
it is true. if you are using a bootloader. you can use a free sector for eeprom.

closed 2022-11-29 by nimaltd