nimaltd/ee #16: STM32F103 ee_format clears 0'th page

remrol, 2022-08-15:
On STM32F1 architecture _EE_PAGE_OR_SECTOR is PAGE, then ee_format instruction:
flashErase.PageAddress = _EE_USE_FLASH_PAGE_OR_SECTOR; 
wrongly assigns page number instead of page address and leads to clearing 0'th bank. Corrected line should be:
flashErase.PageAddress = _EE_ADDR_INUSE;

nimaltd, 2022-08-16:
do you check it? I think I checked before and it works fine.... I can not remember now

remrol, 2022-08-16:
Yes this is the problem which I found just now on STM32F103CBT6 using STM32CubeIDE Version: 1.10.1 HAL 1.8.4 
Original code overwrites first FLASH page and the MCU stops working, need to re-program MCU.
After modification everything works fine.
Of course it will be great if someone check this issue.

nimaltd, 2022-08-16:
thanks for reporting..

nimaltd, 2022-08-16:
please check again. I fixed it

remrol, 2022-08-16:
Works fine thank you, can be closed

closed 2022-08-16 by nimaltd