nimaltd/ee #17: Problem withe the lib for the MCU STM32F411RET6

Alex021488, 2022-11-24:
Hello nimaltd

I have problems with the ee.c libary.
I use the MCU STM32F411RET6 with STM32Cuben IDE. 
The troubleshoot on the debug window show me:

Break at address "0xfffffffe" with no debug information available, or outside of program code.

Can you help me on Microsoft Teams or Skype?

I will pay you for the support. 

Best regards 
Alex Dechand
My e Mail adress: alex.dechand@datalink.ch

nimaltd, 2022-11-25:
Hello. could you send me your project?

Alex021488, 2022-11-25:
[](url)
[Elektrolyse Funktionsmuster.zip](https://github.com/nimaltd/ee/files/10091689/Elektrolyse.Funktionsmuster.zip)

Hello Nimaltd

Thank you very much for your help. 

Attached the project.

nimaltd, 2022-11-29:
I think it is solved now. please check again

closed 2022-11-29 by nimaltd