nimaltd/ee #20: Support for more common u599 (4mb version)

NMSTEC, 2023-08-25:
It's different enough to cause issues if running with u575/u585 definitions

NMSTEC, 2023-08-25:
Function FLASH_EraseInitTypeDef uses page instead PageAddress

nimaltd, 2023-08-26:
Hello. I dont have that chip. if you can add your configuration, please send me and i will add it into library. thanks

NMSTEC, 2023-09-05:
I had to modify the hell out of the code for my own purpose, but it worked. I will not PR it as it completely destroys the original code.
```
//############################################################
bool ee_format(bool keepRamData)
{
  uint32_t error;
  HAL_FLASH_Unlock();
#if _EE_ICACHE_CTRL == 1
  HAL_ICACHE_Disable();
#endif
  FLASH_EraseInitTypeDef flashErase;

	flashErase.NbPages = 1;
  flashErase.Page = _EE_ADDR_INUSE;
  flashErase.TypeErase = FLASH_TYPEERASE_MASSERASE;
  flashErase.Banks = FLASH_BANK_2;

  if (HAL_FLASHEx_Erase(&flashErase, &error) == HAL_OK)
  {
    HAL_FLASH_Lock();
    if (error != 0xFFFFFFFF)
    {
#if _EE_ICACHE_CTRL == 1
    HAL_ICACHE_Enable();
#endif
    	return false;
    }
    else
    {

#if _EE_ICACHE_CTRL == 1
      HAL_ICACHE_Enable();
#endif
      return true;
    }
  }
  HAL_FLASH_Lock();
#if _EE_ICACHE_CTRL == 1
  HAL_ICACHE_Enable();
#endif
  return false;
}

//#######################################################
bool ee_read16(uint32_t startVirtualAddress, uint32_t len, int *data)
{
  // Adjust length to account for the size of an int
  if ((startVirtualAddress + len * sizeof(int)) > _EE_SIZE)
    return false;

  uint32_t intPerQuadWord = 16 / sizeof(int); // A QuadWord is 16 bytes.
  uint8_t QuadWord[16];

  for (uint32_t i = 0; i < len; i += intPerQuadWord)
  {
    // Read a QuadWord (16 bytes) from the flash memory
    uint32_t address = (i * sizeof(int) + startVirtualAddress) + _EE_ADDR_INUSE;
    memcpy(QuadWord, (void *) address, 16);

    for (uint32_t j = 0; j < intPerQuadWord; ++j)
    {
      if (i + j < len && data != NULL)
      {
        memcpy(&data[i + j], &QuadWord[j * sizeof(int)], sizeof(int));
      }
    }

  }

  return true;
}

//######################################################
bool ee_write16(uint32_t startVirtualAddress, uint32_t len, int *data)
{
  if ((startVirtualAddress + len * sizeof(int)) > _EE_SIZE)
    return false;
  if (data == NULL)
    return false;

  HAL_FLASH_Unlock();
#if _EE_ICACHE_CTRL == 1
  HAL_ICACHE_Disable();
#endif

  uint8_t QuadWord[16];

  for (uint32_t i = 0; i < len; i++)
  {
    // Initialize QuadWord with zeros
    memset(QuadWord, 0, sizeof(QuadWord));

    // Copy the int data to the beginning of QuadWord
    memcpy(&QuadWord[0], &data[i], sizeof(int));

    uint32_t flash_address = (i * sizeof(int) + startVirtualAddress) + _EE_ADDR_INUSE;

    if (HAL_FLASH_Program(FLASH_TYPEPROGRAM_QUADWORD, flash_address, (uint32_t)(&QuadWord[0])) != HAL_OK)
    {
      HAL_FLASH_Lock();
#if _EE_ICACHE_CTRL == 1
      HAL_ICACHE_Enable();
#endif
      return false;
    }
  }

  HAL_FLASH_Lock();
#if _EE_ICACHE_CTRL == 1
  HAL_ICACHE_Enable();
#endif

  return true;
}
```
```#ifndef __EECONFIG_H
#define __EECONFIG_H

#define   _EE_USE_FLASH_PAGE_OR_SECTOR              (500) //126 == 0x571d0 250 = 1f4000 500 = 3e8000
#define   _EE_USE_RAM_BYTE                          (1024)
#define   _EE_VOLTAGE                               FLASH_VOLTAGE_RANGE_3 //  use in some devices

#endif
```

NMSTEC, 2023-09-05:
My version will write whatever value is inserted into it, and pad it with 00s if its not big enough,
To use it     
```int toWriteBatch6[] = {Current_Var.1, Current_Var.2, Current_Var.3, Current_Var.4}; //uint16_t
    ee_format(false); 
    ee_write16(0x00, 4, toWriteBatch1);```
    
    I was never able to properly figure out how to delete sectors/pages, so I opted to wipe the whole bank 2. If someone figures it out, I'd be happy to add it.

nimaltd, 2023-09-09:
@NMSTEC I added the configuration. please test it. thanks

NMSTEC, 2023-09-09:
Honestly I rewrote your library nearly fully, and that won't work. U599
only supports burst or quad word.

When I get to work on Monday I'll send you my version. However it's
completely changed from yours. Added quad word padding, uint32_t input, and
array writing.

nimaltd, 2024-03-28:
I have updated the LIB. please try again for any devices you have.