nimaltd/ee #21: Build failed for STM32 L476: "PageAddress" member for FLASH_EraseInitTypeDef not applicable

Uxion, 2023-09-18:
https://github.com/nimaltd/ee/blob/d9f9c7d7ac48163c2c69e4359703ae5de9669f7f/ee.c#L245C19-L245C19

Line 245 of ee.c   " flashErase.PageAddress = _EE_ADDR_INUSE; "

References FLASH_EraseInitTypeDef in stm32l4xx_hal_flash.h driver.

The L4 driver's FLASH_EraseInitTypeDef is missing "PageAddress" member.

typedef struct
{
  uint32_t TypeErase;   /*!< Mass erase or page erase.
                             This parameter can be a value of @ref FLASH_Type_Erase */
  uint32_t Banks;       /*!< Select bank to erase.
                             This parameter must be a value of @ref FLASH_Banks
                             (FLASH_BANK_BOTH should be used only for mass erase) */
  uint32_t Page;        /*!< Initial Flash page to erase when page erase is disabled
                             This parameter must be a value between 0 and (max number of pages in the bank - 1)
                             (eg : 255 for 1MB dual bank) */
  uint32_t NbPages;     /*!< Number of pages to be erased.
                             This parameter must be a value between 1 and (max number of pages in the bank - value of initial page)*/
} FLASH_EraseInitTypeDef;

nimaltd, 2023-10-08:
Please ensure your Selected MCU is Correct. It should be works
'''
#elif _EE_PAGE_OR_SECTOR == PAGE_NUM
  flashErase.NbPages = 1;
  flashErase.Page = _EE_USE_FLASH_PAGE_OR_SECTOR;
  flashErase.TypeErase = FLASH_TYPEERASE_PAGES;
#endif
'''

nimaltd, 2024-03-28:
I have updated the LIB. please try again for any devices you have.