nimaltd/ee #22: bUILD FAILED WITH TONNES OF ERROR

sunilkumar105, 2023-11-28:
[image]

nimaltd, 2023-11-28:
Hello. Your mcu is not in list. Please add it with correct configuration.

sunilkumar105, 2023-11-28:
Hii nimltd
i tried adding our own configuration and studied the architecture of flash of stm32, but still i am missing something as my code is having some issue most probably due to bad configuration of address and it is crashing into hardfault and not getting programmed after reading data from that specific sector untill i erase the wholee flash with stm32 cube programmer software

i am using **STM32F410RB ** with a flash size of **128KILOBYTES**  and **RAM** OF SIZE **32KILOBYTES**
As per the referance manual here is its memory organisation
[image]

as per my memory useage it is only taking 10kb 
[image]
so i can safely use seector 3 
here is my config.h
[image]

Also there was no coorect configuration of my board in ee.c file so i added 
[image]
**QUESTION 1**  IS THE FIRST PARAMETER #define   _EE_SIZE               (1024*16) IS CORRECT? I dont know ehether it is right or wrong but i guessed it to be the size of one page or one sector right ?? If not then please reply with the correct answer

so once i read the flash data from it, after that whenever i try to put it in deebug mode or try to program i keep getting this error
Do u have any clue about it? 
Thanks a lot

nimaltd, 2023-11-28:
@sunilkumar105 Hi, My suggestion: Use the last sector(SEC4, 64KB). it is better. you can use others for your code.

watch this configuratio. i think it like your mcu. 

#if defined(STM32F411xE)
#define   _EE_SIZE              (1024 * 128)
#define   _EE_ADDR_INUSE        (((uint32_t)0x08020000) | (_EE_SIZE*(_EE_USE_FLASH_PAGE_OR_SECTOR - 5)))
#define   _EE_FLASH_BANK        FLASH_BANK_1
#define   _EE_VOLTAGE_RANGE     _EE_VOLTAGE
#define   _EE_PAGE_OR_SECTOR    SECTOR

try with:

#if defined(STM32F410xB)
#define   _EE_SIZE              (1024 * 64)
#define   _EE_ADDR_INUSE        (((uint32_t)0x08020000) | (_EE_SIZE*(_EE_USE_FLASH_PAGE_OR_SECTOR - 4)))
#define   _EE_FLASH_BANK        FLASH_BANK_1
#define   _EE_VOLTAGE_RANGE     _EE_VOLTAGE
#define   _EE_PAGE_OR_SECTOR    SECTOR

and config:
#define   _EE_USE_FLASH_PAGE_OR_SECTOR              (4)

sunilkumar105, 2023-11-28:
sir i tried the above configuration,  now ee_init(); function is taking me to hardfault

nimaltd, 2023-11-28:
@sunilkumar105 sorry. try 0x08010000

nimaltd, 2023-11-28:
#if defined(STM32F410xB)
#define _EE_SIZE (1024 * 64)
#define _EE_ADDR_INUSE (((uint32_t)0x08010000) | (_EE_SIZE*(_EE_USE_FLASH_PAGE_OR_SECTOR - 4)))
#define _EE_FLASH_BANK FLASH_BANK_1
#define _EE_VOLTAGE_RANGE _EE_VOLTAGE
#define _EE_PAGE_OR_SECTOR SECTOR

and config:
#define _EE_USE_FLASH_PAGE_OR_SECTOR (4)

sunilkumar105, 2023-11-28:
Again the issue is same
[image]

sunilkumar105, 2023-11-28:
As per your suggestion u suggested to use #define _EE_ADDR_INUSE (((uint32_t)0x08010000) | (_EE_SIZE*(_EE_USE_FLASH_PAGE_OR_SECTOR - 4)))

but _EE_USE_FLASH_PAGE_OR_SECTOR - 4 will always result in 0, hence making  (_EE_SIZE*(_EE_USE_FLASH_PAGE_OR_SECTOR - 4))) == 0 so _EE_ADDR_INUSE  will always equal to 0x08010000. But we are using sector 4 whouse starting address is 0x08011000

[image]

nimaltd, 2023-11-28:
yes. i see. use 0x0811000

sunilkumar105, 2023-11-29:
but why?  (0X8010000 ) using this address we are pointing to a address of sector 3 and we are defining that we are working with sector 4?

nimaltd, 2023-11-29:
Please Try with that and tell me what is happen