nimaltd/ee #23: data not written on stm32f103c8t6?

TecDroiD, 2023-12-10:
Hi,

i'm currently trying to write and read a data structure into eeprom which looks basically like this

`
typedef struct s_machinecfg {
  uint8_t version,
  uint16_t datax,
  uint16_t datay,
 ..
} t_machinecfg, *p_machinecfg;

typedef struct s_machine {
  t_machinecfg config;
  ..
} t_machine, *p_machine;
...

void read_config() {
ee_read(0, sizeof(t_machinecfg), &self.config);
if (self.config.version != stdconfig.version) {
  memcpy(&self.config, &std_config, sizeof(t_machinecfg));
  write_config();
}
}
void write_config() {
  ee_format(0);
  ee_WriteToRAM(0, sizeof(t_machinecfg), &self.config);
  ee_commit();
}

...
t_machine machine;

ee_init();
read_config();
 
`

later, I want to alter some values in my configuration and save it using  write_config() but every time I restart the controller, I get the unaltered data. If I change my stdconfig in the program, former data is displayed, so first write must work.. but why doesn't the second one do?

Do you have any idea what my problem could be?

my eeConfig.h looks this way:
`
#ifndef __EECONFIG_H
#define __EECONFIG_H

#define   _EE_USE_FLASH_PAGE_OR_SECTOR              (63)
#define   _EE_USE_RAM_BYTE                          (1024)
#define   _EE_VOLTAGE                               FLASH_VOLTAGE_RANGE_3 //  use in some devices
#endif
`

nimaltd, 2023-12-12:
hi. because you should not use format

TecDroiD, 2023-12-13:
Well I just tried format it didn't work before. Removed format now but same effect.

nimaltd, 2023-12-13:
And remove 
if (self.config.version != stdconfig.version) {
memcpy(&self.config, &std_config, sizeof(t_machinecfg));
And try again. And debug step by step to see what is happening. It seems ok.

TecDroiD, 2023-12-13:
i debuged without my autowrite mechanics and it seems that after reading the first byte is 0 instead 0x10..

nimaltd, 2023-12-13:
Let me a while to check.

nimaltd, 2023-12-13:
@TecDroiD I have updated "ee.c". please check again and  tell me the result. thanks

nimaltd, 2023-12-16:
did you check? @TecDroiD

TecDroiD, 2023-12-16:
sorry, was off for a few days. first short experiment doesn't look promising but I promise to do a complete debug today.

TecDroiD, 2023-12-20:
someone broke my blue pill. got a new one now.. and I cleaned the project twice between several debuggings and now it seems to work using uint8_t as first entry in my struct.
may I ask a stupid question? in ee_read you do read the data using *data = ... and increase the pointer after. then you write to ram buffer using ee_ram[i]. Is there any reason I can't see why you didn't use data[i], too?

nimaltd, 2023-12-23:
I am so glad it works. no reason :)

nimaltd, 2024-03-28:
I have updated the LIB. please try again for any devices you have.