nimaltd/ee #24: STM32H7

NicoEFI, 2024-03-04:
Hello
Would it be possible to modify the library to use it with an STM32H757XIH6, it has a flash of up to 2Mb on 2 banks, 8 sectors per bank and 128kb between two sectors. For each recording we must do it on 32bits even to write a single byte.
I need it to use with a Riverdi screen.
If you could port the library that would be great.
[https://www.st.com/content/ccc/resource/training/technical/product_training/group0/1f/4f/b7/85/91/e0/4d/2f/STM32H7-Memory-Flash_FLASH/files/STM32H7-Memory-Flash_FLASH.pdf/_jcr_content/translations/en.STM32H7-Memory-Flash_FLASH.pdf](url)

[https://riverdi.com/fr/product/7-inch-lcd-display-capacitive-touch-panel-air-bonding-atouch-frame-stm32h7-rvt70hssfwca0](url)

nimaltd, 2024-03-09:
Hello. it seems stm32f407, you should use #define   _EE_PAGE_OR_SECTOR    SECTOR
and modify the address and add 8word like U5 series ( 4word is FLASH_TYPEPROGRAM_QUADWORD)

NicoEFI, 2024-03-10:
Thanks for your answer
If I configure like this, is it good?

[image]
[image]

nimaltd, 2024-03-17:
it seems ok. did you try? did you add FLASH_TYPEPROGRAM_QUADWORD and edit writing array?

nimaltd, 2024-03-28:
I have updated the LIB. please try again for any device you have. @NicoEFI