nimaltd/ee #27: "Flash Word" might be 128-bit in some STM32H7 series

ElluIFX, 2024-04-13:
I am using the new version of the ee library in my STM32H7B0VBT6 and I noticed that it writes only 16 bytes per operation instead of the expected 32 bytes.
As I mentioned in my title, after checking the datasheet, it says the "Flash Word" is 128bit on my H7B0, so I think more "#if" is required for the writing function for H7.
But I'm not sure if all single-bank H7 follow this special case, it seems that most answers think the flash word is exactly 256-bit for all H7.
Thanks, nice lib, the datasheet I mentioned is as below:
[image]

nimaltd, 2024-04-14:
Hi. thanks for your report. i am checking now.

nimaltd, 2024-04-14:
It has been updated. please check again

closed 2024-04-14 by nimaltd

ElluIFX, 2024-04-14:
Work great, thanks @nimaltd

nimaltd, 2024-04-14:
Your welcome