nimaltd/ee #28: BUILD FAILED: Linking error in '.ARM.extab' section

yasir-shahzad, 2024-06-07:
My project encountered this issue when I included this library; I'm using the STM32G030F6P6 microcontroller.

arm-none-eabi-gcc -o "Nau7802_485.elf" @"objects.list"   -mcpu=cortex-m0plus -T"C:\Users\HP\STM32CubeIDE\workspace_1.12.1\Nau7802_485\STM32G030F6PX_FLASH.ld" --specs=nosys.specs -Wl,-Map="Nau7802_485.map" -Wl,--gc-sections -static --specs=nano.specs -mfloat-abi=soft -mthumb -Wl,--start-group -lc -lm -Wl,--end-group
c:\st\stm32cubeide_1.12.1\stm32cubeide\plugins\com.st.stm32cube.ide.mcu.externaltools.gnu-tools-for-stm32.10.3-2021.10.win32_1.0.200.202301161003\tools\arm-none-eabi\bin\ld.exe:C:\Users\HP\STM32CubeIDE\workspace_1.12.1\Nau7802_485\STM32G030F6PX_FLASH.ld:88: non constant or forward reference address expression for section .ARM.extab
collect2.exe: error: ld returned 1 exit status
make: *** [makefile:64: Nau7802_485.elf] Error 1
"make -j8 all" terminated with exit code 2. Build might be incomplete.

nimaltd, 2024-06-07:
Please attach your project

yasir-shahzad, 2024-06-07:
[Nau7802_485.zip](https://github.com/user-attachments/files/15721968/Nau7802_485.zip)

yasir-shahzad, 2024-06-08:
Hi @nimaltd, what's the status on the linking error? Has it been resolved?

nimaltd, 2024-06-09:
[g0pp.zip](https://github.com/user-attachments/files/15752000/g0pp.zip)
i could not understand your project. but it works with cpp and G030F6

closed 2024-06-09 by nimaltd