nimaltd/ee #30: STM32F429IIT issue

rrouterr, 2024-07-20:
I have a problem with this library on the STM32F429IIT processor. When trying to erase the last, 23rd sector, nothing happens (the erasure does not occur). So far, I've discovered that the problem lies in calculating the last sector. When I replace:

`#else
    flashErase.TypeErase = FLASH_TYPEERASE_SECTORS;
    flashErase.Sector = EE_PAGE_SECTOR + FLASH_F4_OFFSET;
    flashErase.NbSectors = 1;
#endif
#ifdef EE_BANK_SELECT
    flashErase.Banks = EE_BANK_SELECT;`
    
    with
    
    `#else
    flashErase.TypeErase = FLASH_TYPEERASE_SECTORS;
    flashErase.Sector = 23;  // EE_PAGE_SECTOR + FLASH_F4_OFFSET;
    flashErase.NbSectors = 1;
#endif
#ifdef EE_BANK_SELECT
    // flashErase.Banks = EE_BANK_SELECT;`
    
    the last, 23rd sector is erased, and the write operation can proceed correctly.

nimaltd, 2024-07-21:
hello. can you debug it please and tell me the value of FLASH_SIZE and EE_PAGE_SECTOR

rrouterr, 2024-07-21:
Thank you for your answer. In debug:

flashErase.Banks = 2

#define EE_PAGE_SECTOR ((FLASH_SIZE / EE_SIZE / 2) - 1)
((((((uint32_t)(*((uint16_t *)0x1FFF7A22UL)) & (0xFFFFU))) * 1024) / 0x20000 / 2) - 1)

nimaltd, 2024-07-21:
i mean actual value,  integer
FLASH_SIZE 
EE_PAGE_SECTOR
printf("%d, %d", FLASH_SIZE , EE_PAGE_SECTOR)

rrouterr, 2024-07-21:
yes sorry, the correct answer is:
FLASH_SIZE = 0x200000
EE_PAGE_SECTOR = 7

nimaltd, 2024-08-16:
@rrouterr  I have updated the repository. please try again.