nimaltd/ee #32: EE Library should be set manually for your MCU!

Aydin-M, 2024-09-17:
میکرو من STM32f446 است دستی کتابخونه رو ادد کردم از کیوب نمیاد به بخش select component بعد از ادد کردن و نصب کتابخونه 
الان ارور EE Library should be set manually for your MCU! رو دارم وقتی دستی ادد کردم کتابخونه رو به کیل

nimaltd, 2024-09-17:
۱. کیوب رو آپدیت کنید.
۲. فایل کانفیگ رو تنظیم کنید

Aydin-M, 2024-09-17:
کیوب اپدیت شد 
تو فایل کانفیک کانفیک منوال true شد الان بعد از نوشتن دستور رایت کل میکرو از کار میوفته 
تو تنظیمات EE_SELECTED_ADDRESS رو باید چند بزارم ؟
0x0807F000  هم گذاشتم البته باز کار نمیکنه و کل کد از کار میوفته

/*---------- EE_MANUAL_CONFIG  -----------*/
#define EE_MANUAL_CONFIG      true

/*---------- EE_SELECTED_PAGE_SECTOR_NUMBER  -----------*/
#define EE_SELECTED_PAGE_SECTOR_NUMBER      16

/*---------- EE_SELECTED_PAGE_SECTOR_SIZE  -----------*/
#define EE_SELECTED_PAGE_SECTOR_SIZE      EE_PAGE_SECTOR_SIZE_1K

/*---------- EE_SELECTED_BANK  -----------*/
#define EE_SELECTED_BANK      FLASH_BANK_1

/*---------- EE_SELECTED_ADDRESS  -----------*/
#define EE_SELECTED_ADDRESS      0x08000000

nimaltd, 2024-09-17:
ادرس اول سکتوری که انتخاب کردید. شماره سکتور و حجمشو در دیتاشیت میکرو ببینید. همه تنظیمات اشتباه است.

Aydin-M, 2024-09-17:
ممنون 
تنظیمات به این مقادیر تغییر یافت :
/*---------- EE_MANUAL_CONFIG  -----------*/
#define EE_MANUAL_CONFIG      true

/*---------- EE_SELECTED_PAGE_SECTOR_NUMBER  -----------*/
#define EE_SELECTED_PAGE_SECTOR_NUMBER      7

/*---------- EE_SELECTED_PAGE_SECTOR_SIZE  -----------*/
#define EE_SELECTED_PAGE_SECTOR_SIZE      EE_PAGE_SECTOR_SIZE_128K

/*---------- EE_SELECTED_BANK  -----------*/
#define EE_SELECTED_BANK      FLASH_BANK_1

/*---------- EE_SELECTED_ADDRESS  -----------*/
#define EE_SELECTED_ADDRESS      0x08060000

عکس دیتاشیت :
[image]

الان در کل درسته یا بهتره باز تغییر بدم چون ادرس اول سکتور نهایی رو انتخاب کردم با شماره سکتور ؟

closed 2024-09-17 by nimaltd