nimaltd/ee #34: Compilation error in STM32L053

rajdeep-engineer, 2024-11-15:
In source file "ee.c", line number 227 of version 3.2.1, this line gives an error:-

flashErase.Page = eeHandle.PageSectorNumber;

I changed it to the following and it compiles now:-
flashErase.PageAddress = eeHandle.PageSectorNumber;

nimaltd, 2024-11-15:
Thanks for reporting. I will fix it

rajdeep-engineer, 2024-11-15:
Even though it compiled with the change I made, it doesn't report correctly.

In the screenshot, I stored uint32_t data of 8865, but what I get on reading is a different value. I think it is not selecting the last page of flash automatically. please help.
[image]

nimaltd, 2024-11-15:
Please change #define EE_ERASE EE_ERASE_PAGE_ADDRESS and try again.

rajdeep-engineer, 2024-11-16:
now the value read is always 0
[image]

nimaltd, 2024-11-16:
By a programmer, read flash and check the latest page address. 
And in debug, see the ee struct and tell me values

rajdeep-engineer, 2024-11-16:
It's such a silly thing. I realized that STM32L0 and STM32L1 series microcontrollers have inbuilt EEPROM, so there is no need to emulate EEPROM using Flash memory. We can simply read-write to EEPROM using:-

void EEPROM_Write(uint32_t address, uint32_t data) {
    HAL_FLASHEx_DATAEEPROM_Unlock();
    HAL_FLASHEx_DATAEEPROM_Program(FLASH_TYPEPROGRAMDATA_WORD, address, data);
    HAL_FLASHEx_DATAEEPROM_Lock();
}

uint32_t EEPROM_Read(uint32_t address) {
    return *(uint32_t*)address;
}

closed 2024-11-16 by nimaltd