nimaltd/ee #37: CubeMX cannot install the package

SedonD, 2025-04-27:
Hi, could you look at this question for us and confirm that it is ok with H503....
https://www.eevblog.com/forum/testgear/10mhz-gpsdo-by-bh3sap/msg5894012/#msg5894012

Thank you. DG

closed 2025-04-28 by nimaltd

SedonD, 2025-04-28:
Hi,

Thanks for the reply, CubeMX is the latest version and neither Linux nor Windows installed CubeMX can use the url because of this error msg.
We were able to add Nimaltd to CubeIDE without any problems. see today's posts on eevblog, THX.

[image]

SedonD, 2025-04-28:
tried this 2 minutes ago

[image]

nimaltd, 2025-04-28:
Please try https://github.com/nimaltd/STM32-PACK/raw/main/NimaLTD.pidx

SedonD, 2025-04-28:
it doesn't work either....

[image]

nimaltd, 2025-04-28:
Did you try this link on cubemxide?

SedonD, 2025-04-28:
now, only on Linux with IDE 1.170.0 - this link is not ok

[image]

SedonD, 2025-04-28:
tomorrow I'll update this IDE and try again, right now have total blackout in portugal and spain and I'm working from solar system :-)

nimaltd, 2025-04-28:
Oh oh. I heard. Ok . Please try and tell me.

SedonD, 2025-04-29:
Hi,
Okay, here's the situation:

under win10 with CubeIDE  updated to latest 1.18.1 this link is **OK**....
https://github.com/nimaltd/STM32-PACK/raw/main/NimaLTD.pidx

[image]

under win10 with CubeIDE  updated to latest 1.18.1 this link is **NOT OK**....
https://github.com/nimaltd/STM32-PACK/raw/main/EE/NimaLTD.I-CUBE-EE.pdsc

[image]

nimaltd, 2025-04-29:
Really i have no idea. 🤔

SedonD, 2025-04-29:
and in the end the IDE and MX libraries look like this :-)   ----    this is very strange for me too, but the point is that we now have what we wanted :-)

[image]